newaetech/ChipSHOUTER #7: problem with sensor read and triggersafe feature

h0rac, 2020-03-25:
I have similar problem, but not after 5s.. after 15-60 min of run

```
ChipSHOUTER xformer: 51, MOSFET: 44, diode: 35  - temperature 
ChipSHOUTER voltage measured: 457, set: 470
ChipSHOUTER absent time after trigger: 1
ChipSHOUTER glitched 73 times, runs: 00:17:25
Faults current:  [], latched:  ['fault_temp_sensor'] ['fault_temp_sensor']
```

Also after that when you read temp values they are all -1 or 65535
Only device reset clear them

ChipSHOUTER xformer: 65535, MOSFET: 65535, diode: 65535  - temperature 
This is a bug I think. This values should never be overshoot like that. When reading this value it goes in fault state. It is also possible to Arm with this failure values. If ```trigger_safe``` is not set you can still arm.. but only reset of device clear them

Then when you enable trigger_safe and ARM with xformer: 65535, MOSFET: 65535, diode: 65535 ChipSHOUTER goes to Fault state. This should prove that this is software bug

Of course you can see this -1 values in normal UART menu, so it is not python API problem

h0rac, 2021-10-31:
This issue was solved in 1.8.7 but firmware 2.0.1 is having it again

rtborg, 2022-02-23:
I can confirm that firmware 2.0.1 is having that problem. Is there a possible workaround?

ankoman, 2022-03-09:
I have the same problem

alex-dewar, 2022-03-10:
We've got a new firmware out that should fix the temperature sensor issue entirely, meaning you won't have to use trigger_safe at all. You can update via instructions at https://github.com/newaetech/ChipSHOUTER/blob/master/documentation/Firmware%20Update.md.

Sk3pper, 2024-11-26:
These days, I'm trying to set up a hardware trigger. I have a simple question because I’m a bit confused: when the trigger is activated, should the beep sound be heard, or does the Chipshouter bypass it? I just want to know if everything is working correctly.

alex-dewar, 2025-02-05:
> These days, I'm trying to set up a hardware trigger. I have a simple question because I’m a bit confused: when the trigger is activated, should the beep sound be heard, or does the Chipshouter bypass it? I just want to know if everything is working correctly.

Yeah, I believe that's the case.