newaetech/ChipSHOUTER #9: Unable to bypass fault exception

idonttest, 2021-02-24:
I am using ChipSHOUTER with an external trigger circuit(fpga) connected to my target.  

After few runs, I end up getting the fault["fault_high_voltage", "fault_trigger_glitch"]

I am suspecting the high voltage fault is because of constantly changing the capacitor voltage to control the amplitude of the glitch and second one is because of fault which disarms and glitch still continues.  Please advice on any alternative on controlling the glitch power ?

Even if i call the cmd_clear_fault after catching the latched exception(something similar in the usermanual for reset exception. i keep getting the same exception even after power cycling.

Is there any firmware revision which can disable this high voltage and trigger glitch fault? 

I even tried talking to the serial directly to control it.  but the shouter keeps resetting or more faults. but operation was way  faster than the API.

idonttest, 2021-02-24:
bummer! i think it is already listed in the issues.

colinoflynn, 2021-02-24:
We're still investigating a few of these - when you say it's latched do you mean now it's in the "hv error" state and won't clear at all?

Or does it work for a few pulses and then happen?

What's the firmware version of your device? We have some updates (and are working on a few new issues), the FW are per-device so we have to build it for the boardid.

idonttest, 2021-02-24:
HI Colin, 

I tweak a bit on the code after posting the issue and sorted most of issue and paid a lot of time delay to cool down the caps. 

Issue is because of diff between set/measured V and the fault generated is not cleared even after sending the clr_fault.  I need to power cycle it to make it work again.

colinoflynn, 2021-02-24:
Got it! So if it does clear and then comes back after a few pulses this is likely a firmware fixable issue - we adjusted the "allowable mismatch" range on newer builds, as we had just empirically set some limit between "set and measured". During discharge the measured is somewhat off, and if you are changing the settings a lot it doesn't catch up.

The only "hard" errors we've seen will fault immediately when you attempt to arm (hard meaning we had to swap the unit).

We also changed the way the parameter storage is done - on older FW the EEPROM is written too often (see https://github.com/newaetech/ChipSHOUTER/blob/master/documentation/ERRATA_NAEER02.md). This can cause weird fault modes.

@alex-dewar has been building them on request as we didn't have an easy way to post them yet - easiest solution is to email support@newae.com, or post your board ID here if you prefer (or on discord https://discord.gg/chipwhisperer as can easily send DM). This includes some debug features to print the actual voltage delta which can make it more obvious what's going on.

closed 2021-03-05 by idonttest