newaetech/ChipSHOUTER #10: Serial interface usage issues

lemuellew, 2022-07-21:
I use the mobaxterm to connect to the usb interface, input no echo, and output garbled characters.
[image]

lemuellew, 2022-07-22:
I switched to picocom and had the same problem.

lemuellew, 2022-07-22:
After 'implicit Cr in every lf' is enabled, the output becomes normal, but the input does not echo
[image]

colinoflynn, 2022-07-22:
iirc echo isn't enabled, we also assume local echo is enabled in the terminal emulator program. This is done partially as it makes scripting using the "raw" serial interface easier [i.e., if you don't want to use our Python interface], since you don't need to process your own echo'd back commands. Most terminal emulators support local echo option so it seemed to be the most useful compromise for customers, even though it's not perfect either way!

closed 2022-07-27 by lemuellew