newaetech/ChipSHOUTER #11: Setting pulse width when using hardware trigger from ChipWhisperer-Lite

itewqq, 2022-12-16:
I am using chipwhisperer and chipsohuter jointly and I would like to know how to modify the output pulse width of the chipshouter? 

I have tried to modify the `scope.glitch.repeat` parameter on the chipwhisperer side and the `cs.pulse.width` parameter on the chipshouter side, but it doesn't look like this is working from the oscilloscope.

colinoflynn, 2022-12-16:
What widths are you testing? FYI the pulse width gets highly limited by the physical coil parameters. Page 20/21 has this graph of the [user manual](https://github.com/newaetech/ChipSHOUTER/blob/master/documentation/ChipSHOUTER%20User%20Manual.pdf) that shows the limits you can expect.

[image]
[image]

> I have tried to modify the scope.glitch.repeat parameter on the chipwhisperer side and the cs.pulse.width parameter on the ?  chipshouter side

Using the mode as `glitch.output = enable_only` then modifying `repeat` should give a longer & longer input signal which translates to a longer output. Trying modifying the repeat to extremes as well (e.g., 10 vs 10000).

The `cs.pulse.width` parameter only takes effect if you call the glitch on the ChipSHOUTER API itself, the `cs.pulse.width` is ignored for the external input. The external input is a fully hardware path that basically mirrors the input on the external input (the MCX input on the ChipSHOUTER) to the MOSFET gate itself.

itewqq, 2022-12-17:
Thanks for the fast reply!

>Using the mode as `glitch.output = enable_only` then modifying `repeat` should give a longer & longer input signal which translates to a longer output. Trying modifying the repeat to extremes as well (e.g., 10 vs 10000).

This is very important information and much appreciated. I'll give it a shot :smiling_face_with_tear:

closed 2024-11-25 by itewqq