newaetech/ChipSHOUTER #12: How to set the trigger delay

HoulaiZhang, 2024-12-12:
When I connect the hardware trigger to the trigger of another development board, ChipSHOUTER will pulse as soon as it receives the trigger signal.
Is there a way to control the ChipSHOUTER to delay the pulse at a controllable time after receiving the trigger signal?

alex-dewar, 2024-12-13:
Hi,

The hardware trigger is pretty simple, so there's no way to delay from the ChipShouter itself. Typically we use a ChipWhisperer to delay incoming trigger signals.

HoulaiZhang, 2025-03-05:
> Hi,
> 
> The hardware trigger is pretty simple, so there's no way to delay from the ChipShouter itself. Typically we use a ChipWhisperer to delay incoming trigger signals.

Are there any pictures or documentation about delayed collaboration between chipSHOUTER and chipwhisperer?

alex-dewar, 2025-03-05:
There's no pictures, but this method is described on pages 24-26 of the manual. The delay can be set with `scope.glitch.ext_offset`.